orig74/DroneSimLab #6: PROJECT_FILES_DIR & PROJECT_NAME

orig74, 2017-01-11:
need to think what to do with those environment variables define in the tmux scripts and the run image scripts

closed 2017-01-24 by orig74

ykhedar, 2018-03-29:
Hi, I was trying to run the px4_gazebo demo with the editor. But it seems there are no project files in the project directory

> PROJECT_FILES_DIR=/local/learn/ur4

Does one have to create the game files and put it into this directory or I am missing something?

Thanks for your help. :)

orig74, 2018-03-30:
When you are using the Unreal Engine Editor, the idea is that you create your own virtual environment project yourself and add the dronesimlab components to it with the instructions given by the dronesimlab wiki.