orig74/DroneSimLab #11: the demo is not running on first run after installation

orig74, 2017-03-10:
This might happen because it takes time for the gazebo px4 simulation to compile at first run. After running this demo again it works.

hassanyf, 2020-01-17:
I ran `./runtmux_game_docker.sh`, but just see one windows and the terminal is empty. It's been more than 1 hour.

[image]

orig74, 2020-01-18:
please attach the /tmp/dbuild-ros_image_kinetic.log file.
the docker build for the ros_image_kinetic might have failed.

hassanyf, 2020-01-20:
I dont see any log file there. 

[image]

The build failed for the ros_image_kinetic but then I ran `cd dockers/ros_image_kinetic && ../build –nocache` and it was built successfully

[image]

tomlogan501, 2020-03-11:
Hi,
How do you manage to get the unreal engine ? Should I change the submodule url to somewhere else ?