orig74/DroneSimLab #16: build issue: Unable to find image 'ros_image_kinetic:latest' locally

TheRobotStudio, 2017-07-20:
Hello,

I have an issue during the build on Ubuntu 16.04, at the end I got the error:
```
Unable to find image 'ros_image_kinetic:latest' locally
Pulling repository docker.io/library/ros_image_kinetic
docker: Error: image library/ros_image_kinetic:latest not found.
See 'docker run --help'.
Error:  Error: faild installing ros
```
and my images doesn't show ros:
```
$ sudo docker images
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
<none>              <none>              79fe18f60832        2 hours ago         1.721 GB
<none>              <none>              a26d5a0db4be        2 hours ago         1.975 GB
<none>              <none>              f1e6486e57a9        2 hours ago         2.773 GB
<none>              <none>              3934d690274a        3 hours ago         11.34 GB
<none>              <none>              d857557c1f3f        3 hours ago         4.57 GB
ubuntu              16.04               d355ed3537e9        4 weeks ago         119.2 MB
ubuntu              xenial              d355ed3537e9        4 weeks ago         119.2 MB
ubuntu              14.04               4a2820e686c4        4 weeks ago         188 MB
hello-world         latest              1815c82652c0        5 weeks ago         1.84 kB
ubuntu              xenial-20170417.1   f7b3f317ec73        12 weeks ago        117.3 MB
```
I'm new to dockers, how can I solve this issue ?
Thanks in advance!

danielbarry, 2017-07-20:
Hi there :)

Can we get some information about your system please?

* `uname -r` - Kernel version
* `docker --version` - Docker version
* `sudo lshw` - Hardware
* OS being used

It seems as though some people have issue with out of date docker versions [1]. Let's check the simple stuff before fully debugging.

[[1](https://github.com/openshift/origin/issues/9834)] GitHub

TheRobotStudio, 2017-07-20:
Thanks for your reply. Here are the outputs:
```
$ uname -r
4.4.0-83-generic
$ docker --version
Docker version 1.12.6, build 78d1802
$ sudo lshw
[sudo] password for cyril: 
cyril-z170-a              
    description: Desktop Computer
    product: System Product Name (SKU)
    vendor: System manufacturer
    version: System Version
    serial: System Serial Number
    width: 64 bits
    capabilities: smbios-3.0 dmi-3.0 vsyscall32
    configuration: boot=normal chassis=desktop family=To be filled by O.E.M. sku=SKU uuid=805761CB-3156-BD11-9106-38D547137ED4
  *-core
       description: Motherboard
       product: Z170-A
       vendor: ASUSTeK COMPUTER INC.
       physical id: 0
       version: Rev 1.xx
       serial: 160881908900816
       slot: Default string
     *-firmware
          description: BIOS
          vendor: American Megatrends Inc.
          physical id: 0
          version: 1902
          date: 06/24/2016
          size: 64KiB
          capacity: 15MiB
          capabilities: pci apm upgrade shadowing cdboot bootselect socketedrom edd int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer acpi usb biosbootspecification uefi
     *-cache:0
          description: L1 cache
          physical id: 45
          slot: L1 Cache
          size: 128KiB
          capacity: 128KiB
          capabilities: synchronous internal write-back data
          configuration: level=1
     *-cache:1
          description: L1 cache
          physical id: 46
          slot: L1 Cache
          size: 128KiB
          capacity: 128KiB
          capabilities: synchronous internal write-back instruction
          configuration: level=1
     *-cache:2
          description: L2 cache
          physical id: 47
          slot: L2 Cache
          size: 1MiB
          capacity: 1MiB
          capabilities: synchronous internal write-back unified
          configuration: level=2
     *-cache:3
          description: L3 cache
          physical id: 48
          slot: L3 Cache
          size: 8MiB
          capacity: 8MiB
          capabilities: synchronous internal write-back unified
          configuration: level=3
     *-cpu
          description: CPU
          product: Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz
          vendor: Intel Corp.
[... 619 lines cut ...]
             size: 698GiB (750GB)
             capabilities: partitioned partitioned:dos
             configuration: ansiversion=5 logicalsectorsize=512 sectorsize=4096 signature=3c9460fb
           *-volume
                description: Windows NTFS volume
                physical id: 1
                bus info: scsi@5:0.0.0,1
                logical name: /dev/sdc1
                logical name: /media/windows10_D
                version: 3.1
                serial: 2274f751-785c-734a-aca1-08c92f7b8a55
                size: 698GiB
                capacity: 698GiB
                capabilities: primary ntfs initialized
                configuration: clustersize=4096 created=2016-11-22 16:02:40 filesystem=ntfs label=Cyril modified_by_chkdsk=true mount.fstype=fuseblk mount.options=rw,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096 mounted_on_nt4=true resize_log_file=true state=mounted upgrade_on_mount=true
  *-power UNCLAIMED
       description: To Be Filled By O.E.M.
       product: To Be Filled By O.E.M.
       vendor: To Be Filled By O.E.M.
       physical id: 1
       version: To Be Filled By O.E.M.
       serial: To Be Filled By O.E.M.
       capacity: 32768mWh
```
I'm using Ubuntu 16.04 LTS.

danielbarry, 2017-07-20:
Thanks. All seems good so far, how about: `nvidia-smi `?

TheRobotStudio, 2017-07-20:
Thx, ok here it is:
$ nvidia-smi
Thu Jul 20 20:13:51 2017       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 375.66                 Driver Version: 375.66                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 106...  Off  | 0000:01:00.0      On |                  N/A |
| 36%   54C    P0    31W / 200W |    808MiB /  6069MiB |      4%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID  Type  Process name                               Usage      |
|=============================================================================|
|    0      1464    G   /usr/lib/xorg/Xorg                             491MiB |
|    0      2268    G   compiz                                          24MiB |
|    0      2711    G   /usr/lib/firefox/firefox                       258MiB |
|    0      7168    G   gzserver                                         3MiB |
|    0      7169    G   gzclient                                        28MiB |
+-----------------------------------------------------------------------------+

danielbarry, 2017-07-20:
I just noticed you ran `sudo docker images` earlier. What happens if you run `docker images`?

TheRobotStudio, 2017-07-20:
Yes it's a good point, it gives me an error message:
```
$ docker images
Cannot connect to the Docker daemon. Is the docker daemon running on this host?
```

danielbarry, 2017-07-20:
Yep, so you need to follow these instructions:

https://askubuntu.com/questions/477551/how-can-i-use-docker-without-sudo

TheRobotStudio, 2017-07-21:
Thanks I'll try that.
I was wondering, as I have Ubuntu 16.04 with ROS, Gazebo and Unreal Editor already installed, can I run DroneSimLab without docker ?

orig74, 2017-07-22:
It possible, but we haven't created an example for that kind of scenario. The reason for hiding all behind Docker instances was to avoid network collision in case of more the one robot and to manage version dependencies between all the simulation components for each case scenario e.g. python, ubuntu UR4 ROS SITL, etc. For example, for the drone tracking drone example, we used SITL from Ardupilot which was compiled on ubuntu 14 and tested using Python 2.7.

closed 2017-07-22 by orig74

TheRobotStudio, 2017-07-28:
Ok I understand, the demo now works after connecting my user name to the docker group, thanks!