orig74/DroneSimLab #18: Docker Unreal gives an error pyinit not defined

geevargs, 2017-09-05:
Whenever I start unreal after following procedure for setup.
It works most of the time but sometimes it gives an error 
[image]
[image]

[image]

geevargs, 2017-09-06:
I get this error sometimes when I try the render to texture game mode.

[image]

orig74, 2017-09-06:
For the "pyinit not defined" error. This is usually the case of running in a "stand alone" game mode. Make sure you select the  "viewport mode"  when you hit the play button in the editor. Also hit the stop button before you start the simulation that will make sure that the previous resources are properly cleaned.

orig74, 2017-09-06:
As for the other texture game error. Please send me the steps you did to get this error?

geevargs, 2017-09-06:
I use the selected viewport not standalone viewport and use rendertotexture game mode.
Do you not use any gamemode and just play it directly.

orig74, 2017-09-06:
I never used render to texture game mode.  There are two ways I used to run the simulation:
1. Direct selected viewport Play
2. As a Baked (compiled) game (like some of the demos)

geevargs, 2017-09-06:
I use the direct selected viewport and get the pyint not defined.
[image]

geevargs, 2017-09-06:
Almost fixed...

[image]

geevargs, 2017-09-06:
It was some modfication I did...to the drone ip address...I want to send the udp packets from another ip address 10.0.1.30 with the lat,long,alt,pitch,roll and yaw.

geevargs, 2017-09-06:
And I also noticed that if I dont kill the unreal engine,It causes the Pyint not found error.

closed 2017-09-12 by orig74