orig74/DroneSimLab #19: The two drone ardupilot windows freeze on execution in unreal engine

geevargs, 2017-09-06:
[image]

geevargs, 2017-09-06:
This happens sometimes...it just crashes...
[image]

orig74, 2017-09-07:
The "freezing" part usually happens in the python_proxy.py part if it needs to handle too much communication. You can unmark line 82 which prints the incoming communication and the input position to see if that is the case.

geevargs, 2017-09-07:
You mean unreal_proxy.py....
[image]

orig74, 2017-09-07:
Sorry,
The unreal_proxy.py
and the line to un comment is:
82: #print('-----',positions[drone_index])

geevargs, 2017-09-07:
[image]

geevargs, 2017-09-08:
This is what its printing and the Drone 0 and Drone 1 are not moving in their windows but the data is being received and also I frequently get a crash message like the one in the end.

sending state main loop
----- {'posy': 0.094048, 'type': 'udp_patch', 'pitch': 1.651767, 'yaw': -8.287836, 'posx': 0.279356, 'posz': 5.323888, 'roll': -0.614514}
----- {'posy': 0.090676, 'type': 'udp_patch', 'pitch': 1.653989, 'yaw': -8.287579, 'posx': 0.307425, 'posz': 5.334369, 'roll': -0.648128}
----- {'posy': 0.087318, 'type': 'udp_patch', 'pitch': 1.657492, 'yaw': -8.287355, 'posx': 0.335641, 'posz': 5.34605, 'roll': -0.680035}
----- {'posy': 0.08415, 'type': 'udp_patch', 'pitch': 1.662881, 'yaw': -8.287161, 'posx': 0.364012, 'posz': 5.357872, 'roll': -0.709629}
----- {'posy': 0.081017, 'type': 'udp_patch', 'pitch': 1.669965, 'yaw': -8.287, 'posx': 0.392475, 'posz': 5.370334, 'roll': -0.739087}
----- {'posy': 0.077658, 'type': 'udp_patch', 'pitch': 1.6802, 'yaw': -8.28687, 'posx': 0.421409, 'posz': 5.382857, 'roll': -0.768659}
----- {'posy': 0.074027, 'type': 'udp_patch', 'pitch': 1.693807, 'yaw': -8.286668, 'posx': 0.45067, 'posz': 5.396854, 'roll': -0.796862}
----- {'posy': 0.070013, 'type': 'udp_patch', 'pitch': 1.716764, 'yaw': -8.286597, 'posx': 0.480119, 'posz': 5.412022, 'roll': -0.818582}
----- {'posy': 0.065961, 'type': 'udp_patch', 'pitch': 1.751234, 'yaw': -8.286654, 'posx': 0.509686, 'posz': 5.425475, 'roll': -0.830768}
----- {'posy': 0.061981, 'type': 'udp_patch', 'pitch': 1.794095, 'yaw': -8.286831, 'posx': 0.539439, 'posz': 5.437361, 'roll': -0.833996}
----- {'posy': 0.058058, 'type': 'udp_patch', 'pitch': 1.843849, 'yaw': -8.287008, 'posx': 0.569322, 'posz': 5.44926, 'roll': -0.830769}
----- {'posy': 0.05414, 'type': 'udp_patch', 'pitch': 1.898498, 'yaw': -8.287216, 'posx': 0.599046, 'posz': 5.459823, 'roll': -0.824161}
----- {'posy': 0.128735, 'type': 'udp_patch', 'pitch': 1.925776, 'yaw': -8.246104, 'posx': -0.037096, 'posz': 5.314987, 'roll': -0.301666}
----- {'posy': 0.12474, 'type': 'udp_patch', 'pitch': 1.910038, 'yaw': -8.246314, 'posx': -0.009519, 'posz': 5.323722, 'roll': -0.322527}
----- {'posy': 0.12058, 'type': 'udp_patch', 'pitch': 1.892075, 'yaw': -8.246545, 'posx': 0.018097, 'posz': 5.332512, 'roll': -0.346037}
----- {'posy': 0.116572, 'type': 'udp_patch', 'pitch': 1.871485, 'yaw': -8.246717, 'posx': 0.045794, 'posz': 5.34122, 'roll': -0.369573}
----- {'posy': 0.112441, 'type': 'udp_patch', 'pitch': 1.85026, 'yaw': -8.24682, 'posx': 0.073757, 'posz': 5.351737, 'roll': -0.390786}
----- {'posy': 0.108285, 'type': 'udp_patch', 'pitch': 1.833939, 'yaw': -8.246967, 'posx': 0.101683, 'posz': 5.363452, 'roll': -0.408769}
----- {'posy': 0.104246, 'type': 'udp_patch', 'pitch': 1.820571, 'yaw': -8.247109, 'posx': 0.129706, 'posz': 5.375309, 'roll': -0.422583}
----- {'posy': 0.100389, 'type': 'udp_patch', 'pitch': 1.809648, 'yaw': -8.247136, 'posx': 0.157854, 'posz': 5.387805, 'roll': -0.433524}
----- {'posy': 0.096545, 'type': 'udp_patch', 'pitch': 1.801397, 'yaw': -8.247027, 'posx': 0.185959, 'posz': 5.400362, 'roll': -0.442755}
----- {'posy': 0.092957, 'type': 'udp_patch', 'pitch': 1.793552, 'yaw': -8.246904, 'posx': 0.214175, 'posz': 5.414391, 'roll': -0.452827}
----- {'posy': 0.089881, 'type': 'udp_patch', 'pitch': 1.788728, 'yaw': -8.246799, 'posx': 0.242505, 'posz': 5.429591, 'roll': -0.466114}
----- {'posy': 0.086941, 'type': 'udp_patch', 'pitch': 1.787369, 'yaw': -8.24663, 'posx': 0.271143, 'posz': 5.44308, 'roll': -0.485369}
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655

orig74, 2017-09-08:
Can you identify the line that it fails?
You can do that by just adding more prints in the unreal_proxy.py

geevargs, 2017-09-08:
I put a print in line 85 called print('---------4') and line 99 I called    print('--------- 5')
sending state main loop
sending state main loop
sending state main loop
sending state main loop
sending state main loop
sending state main loop
sending state main loop
sending state main loop
sending state main loop
----- {'posy': 0.024114, 'posz': 76.780342, 'posx': 0.036437, 'type': 'udp_patch', 'pitch': 18.447004, 'roll': -2.7362, 'yaw': -7.907343}
----- {'posy': 0.024485, 'posz': 76.799774, 'posx': 0.036153, 'type': 'udp_patch', 'pitch': 18.467422, 'roll': -2.6883, 'yaw': -7.899324}
----- {'posy': 0.025146, 'posz': 76.81749, 'posx': 0.035421, 'type': 'udp_patch', 'pitch': 18.468803, 'roll': -2.649037, 'yaw': -7.890746}
----- {'posy': 0.025464, 'posz': 76.834061, 'posx': 0.034541, 'type': 'udp_patch', 'pitch': 18.455015, 'roll': -2.614197, 'yaw': -7.88215}
----- {'posy': 0.025538, 'posz': 76.851547, 'posx': 0.033381, 'type': 'udp_patch', 'pitch': 18.427162, 'roll': -2.582943, 'yaw': -7.873595}
----- {'posy': 0.025103, 'posz': 76.868843, 'posx': 0.031697, 'type': 'udp_patch', 'pitch': 18.386402, 'roll': -2.552732, 'yaw': -7.865148}
----- {'posy': 0.024782, 'posz': 76.886238, 'posx': 0.029627, 'type': 'udp_patch', 'pitch': 18.332211, 'roll': -2.522676, 'yaw': -7.856841}
----- {'posy': 0.024459, 'posz': 76.903175, 'posx': 0.027231, 'type': 'udp_patch', 'pitch': 18.262623, 'roll': -2.493108, 'yaw': -7.848665}
----- {'posy': 0.024191, 'posz': 76.918304, 'posx': 0.024809, 'type': 'udp_patch', 'pitch': 18.179474, 'roll': -2.46446, 'yaw': -7.840856}
----- {'posy': 0.02403, 'posz': 76.932274, 'posx': 0.022231, 'type': 'udp_patch', 'pitch': 18.092323, 'roll': -2.431926, 'yaw': -7.833872}
----- {'posy': 0.024064, 'posz': 76.94738, 'posx': 0.019179, 'type': 'udp_patch', 'pitch': 18.001606, 'roll': -2.397241, 'yaw': -7.827748}
----- {'posy': 0.024095, 'posz': 76.962296, 'posx': 0.015589, 'type': 'udp_patch', 'pitch': 17.905718, 'roll': -2.364122, 'yaw': -7.822407}
----- {'posy': 0.024134, 'posz': 76.97718, 'posx': 0.011481, 'type': 'udp_patch', 'pitch': 17.802298, 'roll': -2.335589, 'yaw': -7.81771}
----- {'posy': 0.024198, 'posz': 76.993469, 'posx': 0.006974, 'type': 'udp_patch', 'pitch': 17.688688, 'roll': -2.311797, 'yaw': -7.813409}
----- {'posy': 0.024206, 'posz': 77.013069, 'posx': 0.00208, 'type': 'udp_patch', 'pitch': 17.562399, 'roll': -2.291916, 'yaw': -7.8095}
----- {'posy': 0.024193, 'posz': 77.041534, 'posx': -0.004444, 'type': 'udp_patch', 'pitch': 17.389267, 'roll': -2.263367, 'yaw': -7.805466}
---------4
----- {'posy': 0.044538, 'posz': 76.808243, 'posx': 0.016727, 'type': 'udp_patch', 'pitch': 18.183887, 'roll': -2.547236, 'yaw': -7.937316}
----- {'posy': 0.043229, 'posz': 76.825714, 'posx': 0.017041, 'type': 'udp_patch', 'pitch': 18.243397, 'roll': -2.489088, 'yaw': -7.931499}
----- {'posy': 0.041466, 'posz': 76.846024, 'posx': 0.017525, 'type': 'udp_patch', 'pitch': 18.289824, 'roll': -2.419451, 'yaw': -7.923951}
----- {'posy': 0.039995, 'posz': 76.863419, 'posx': 0.017493, 'type': 'udp_patch', 'pitch': 18.311337, 'roll': -2.363168, 'yaw': -7.917001}
----- {'posy': 0.038708, 'posz': 76.879196, 'posx': 0.017154, 'type': 'udp_patch', 'pitch': 18.31423, 'roll': -2.314387, 'yaw': -7.90994}
----- {'posy': 0.037478, 'posz': 76.898773, 'posx': 0.015934, 'type': 'udp_patch', 'pitch': 18.292383, 'roll': -2.253036, 'yaw': -7.899717}
----- {'posy': 0.036592, 'posz': 76.919289, 'posx': 0.013353, 'type': 'udp_patch', 'pitch': 18.244781, 'roll': -2.197345, 'yaw': -7.889265}
----- {'posy': 0.036031, 'posz': 76.931374, 'posx': 0.011391, 'type': 'udp_patch', 'pitch': 18.202585, 'roll': -2.171805, 'yaw': -7.88334}
----- {'posy': 0.034996, 'posz': 76.950439, 'posx': 0.007538, 'type': 'udp_patch', 'pitch': 18.103865, 'roll': -2.140436, 'yaw': -7.873456}
----- {'posy': 0.034231, 'posz': 76.966652, 'posx': 0.004054, 'type': 'udp_patch', 'pitch': 18.0044, 'roll': -2.123336, 'yaw': -7.865646}
----- {'posy': 0.033582, 'posz': 76.985649, 'posx': 0.000249, 'type': 'udp_patch', 'pitch': 17.885311, 'roll': -2.111038, 'yaw': -7.858045}
----- {'posy': 0.033029, 'posz': 77.006523, 'posx': -0.003598, 'type': 'udp_patch', 'pitch': 17.754181, 'roll': -2.100082, 'yaw': -7.851014}
----- {'posy': 0.032449, 'posz': 77.028137, 'posx': -0.007428, 'type': 'udp_patch', 'pitch': 17.615599, 'roll': -2.090227, 'yaw': -7.844836}
----- {'posy': 0.031723, 'posz': 77.051796, 'posx': -0.010789, 'type': 'udp_patch', 'pitch': 17.474546, 'roll': -2.082578, 'yaw': -7.839696}
----- {'posy': 0.031186, 'posz': 77.077126, 'posx': -0.01375, 'type': 'udp_patch', 'pitch': 17.337275, 'roll': -2.077476, 'yaw': -7.835735}
----- {'posy': 0.030565, 'posz': 77.102669, 'posx': -0.01657, 'type': 'udp_patch', 'pitch': 17.210161, 'roll': -2.077498, 'yaw': -7.832991}
---------4
--------- 5

geevargs, 2017-09-08:
I tried a smaller map that you used...but the drones still don't move.It is more stable.
[image]

orig74, 2017-09-08:
I think the line it fails is 101 (in the original file, please verify by printing before and after that line). This means that it can't get the texture data. If that is the case, we will continue from there. Please also make sure that the texture names (line 10,11) as in the unreal engine.

geevargs, 2017-09-09:
I tested print('----8',new_pos) by putting prints at line 88
[image]

geevargs, 2017-09-09:
I put the print before and after line 101 and I didn't have any problems.
It gets a segmentation fault every time.
[image]

orig74, 2017-09-09:
Hi,
You should see two HELLO, COOL prints because we have two drones. So in the large setup, you probably have only one drone actor or some mismatch in the names. You can add the line:
import pdb;psb.set_trace()
At the beginning of the while loop (line 76), it will freeze the simulation and the engine, but you will able using the keys 'c' for continuing and  'n' for next to identify the exact line it is failing in.
Let me know how it goes.

geevargs, 2017-09-11:
According to my observation,It goes and gets stuck in the the while loop in  line 80 and goes till 84.
[image]

orig74, 2017-09-11:
It's not getting stuck there, it just tries to read from the socket as long as it has something there, and in the debugger, there will always be something there. put the "import pdb..."  line after line  90. That way it will be after it received the locations from the socket.

geevargs, 2017-09-11:
/DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(77)main_loop()
-> socket_sub=drone_subs[drone_index]
(Pdb) c
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(77)main_loop()
-> socket_sub=drone_subs[drone_index]
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(92)main_loop()
-> for drone_index in range(config.n_drones):
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(93)main_loop()
-> #img=cv2.resize(ph.GetTextureData(drone_textures[drone_index]),(1024,1024),cv2.INTER_LINEAR)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(95)main_loop()
-> imgs=[]
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(96)main_loop()
-> pdb.set_trace()
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(98)main_loop()
-> topics.append(config.topic_unreal_drone_rgb_camera%drone_index)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(99)main_loop()
-> imgs.append(img)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(100)main_loop()
-> 
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(102)main_loop()
-> img_down=ph.GetTextureData(drone_textures_down[drone_index])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(103)main_loop()
-> topics.append(config.topic_unreal_drone_rgb_camera%drone_index+b'down')
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(104)main_loop()
-> imgs.append(img_down)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(105)main_loop()
-> 
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(107)main_loop()
-> img_depth=ph.GetTextureData16f(drone_textures_depth[drone_index],channels=[0,1,2,3]) #depth data will be in A componnent
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(108)main_loop()
-> #img_depth=ph.GetTextureData(drone_textures_depth[drone_index],channels=[2]) #depth data will be in red componnent
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(110)main_loop()
-> imgs.append(img_depth)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(111)main_loop()
-> 
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(119)main_loop()
-> for topic,img in zip(topics,imgs):
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(120)main_loop()
-> #socket_pub.send_multipart([topic,pickle.dumps(img,2)])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(123)main_loop()
-> #socket_pub.send_multipart([topic,pickle.dumps(img,-1)])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(120)main_loop()
-> #socket_pub.send_multipart([topic,pickle.dumps(img,2)])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(123)main_loop()
-> #socket_pub.send_multipart([topic,pickle.dumps(img,-1)])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(120)main_loop()
-> #socket_pub.send_multipart([topic,pickle.dumps(img,2)])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(123)main_loop()
[... 61 lines cut ...]
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(81)main_loop()
-> positions[drone_index]=pickle.loads(msg)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(82)main_loop()
-> #print('-----',positions[drone_index])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(80)main_loop()
-> topic, msg = socket_sub.recv_multipart()
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(81)main_loop()
-> positions[drone_index]=pickle.loads(msg)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(82)main_loop()
-> #print('-----',positions[drone_index])
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(80)main_loop()
-> topic, msg = socket_sub.recv_multipart()
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(81)main_loop()
-> positions[drone_index]=pickle.loads(msg)
(Pdb) n
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(82)main_loop()
-> #print('-----',positions[drone_index])
(Pdb)

orig74, 2017-09-11:
I can't figure out why it is jumping from line 93 to 77 (in the second time) it supposed to run this loop twice. Verify that the config.n_drones is equal to 2. you can do it in the debugger. This loop should run twice for each drone.

geevargs, 2017-09-11:
I verified that n_config=2
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(80)main_loop()
-> topic, msg = socket_sub.recv_multipart()
(Pdb) 
> /DroneLab/demos/unreal_proxies/two_drones/unreal_proxy.py(81)main_loop()
-> positions[drone_index]=pickle.loads(msg)
(Pdb) config.n_drones
2
(Pdb)

orig74, 2017-09-11:
Sorry, My mistake, It did run twice I can see from your prints that it did the loop twice. Did you remove the "HELLO, COOL" prints? we should see those prints in the debugger. Just to clarify things, I want to understand if it behaves differently with and without the debugger. Since at the beginning you showed me only one set of   "HELLO, COOL" prints and if the there is no crash in the loop we should see 2.

geevargs, 2017-09-11:
[image]

orig74, 2017-09-11:
I can see that it is running fine in the debugger. If you remove the "import pdb.. " line, it crashes?

geevargs, 2017-09-11:
Yes it gives a segmentation fault...Signal 11.

geevargs, 2017-09-11:
[image]

orig74, 2017-09-11:
The last print screen looks OK. Did you change anything?

geevargs, 2017-09-11:
For the last image I disabled the pdb.set_trace and put a print in line 98 and 106.
The drone 0 and 1 dont move but the sky transitions in the window and the grass moves.

orig74, 2017-09-11:
Can you add this print before line 87 ("ph.SetActorLocation(drone_actor,new_pos)"):
print('-----',drone_index,new_pos)
This is right before I send the new position to unreal engine.
Make sure that it is changed and the values are making sense.

geevargs, 2017-09-12:
[image]

orig74, 2017-09-12:
I can see that the position is changed can you please check that this is not a scale issue?
Look at the values and the relative position between the drones. Does it make sense?
What is the position of each drone in the starting point and in the middle of the scenario?

orig74, 2017-09-12:
Is your Parrot_Drone object defined as Movable?

geevargs, 2017-09-12:
It worked..It was static changed it to movable now it works.
[image]

orig74, 2017-09-12:
great 👍

closed 2017-09-12 by orig74