orig74/DroneSimLab #20: The unreal engine docker crashes on execution

geevargs, 2017-09-29:
I am trying  to use the unreal docker in unreal engine but I keep getting the Signal 11 or segmentation fault error...2 out 3 times i run unreal engine.
I assumed that it was because of the memory input so I built a smaller map similar to the demo but it still persists and is it the unreal engine.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
./run.sh: line 9:  1357 Segmentation fault      (core dumped) Engine/Binaries/Linux/UE4Editor "/DroneLab/MyProject2/MyProject2.uproject" -nocore -project=/DroneLab/MyProject2/MyProject2.uproject
[image]

orig74, 2017-09-29:
Do you have two drones created in the environment?
It looks like it printed the HELLO COOL only once and probably exited on the second drone. you can check that by changing the config.n_drones to 1 instead of 2

geevargs, 2017-09-29:
I am using only one drone the other one I am sending directly through my host.

geevargs, 2017-09-29:
Still crashes frequently....
[image]

orig74, 2017-09-29:
try to run it without sending anything from your host, is it still crashes?

geevargs, 2017-09-29:
It happens when I click play and fails regardless of where I send the data from whether its the SITL image or from my host machine.

geevargs, 2017-09-29:
I tried running it with out any data coming into it from any source and it crashed on play.

orig74, 2017-10-13:
Updated to a newer version of unreal engine.

closed 2017-10-13 by orig74