orig74/DroneSimLab #21: Upgraded to the new Unreal Engine and now not recognizing the pyinit

geevargs, 2017-10-04:
[image]

geevargs, 2017-10-04:
I use the selected viewport when I press play
/DroneLab/MyProject2$./build.sh 
Using clang version '3.8.0' (string), 3 (major), 8 (minor), 0 (patch)
@progress push 5%
@progress 'Generating code...' 0%
@progress 'Generating code...' 67%
@progress 'Generating code...' 100%
@progress pop
Using bundled libc++ standard C++ library.
/bin/sh  -c '/bin/sh "/local/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/FixDependencies.sh" && echo "Dummy" >> "/local/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/UE4Editor.link"'
Target is up to date

geevargs, 2017-10-04:
This is the error I am getting:
2017.10.04-22.10.11:129][142]LogTemp:Warning: called APyServerTickActor::BeginPlay

loading libfile /DroneLab/MyProject2/Binaries/Linux/libUE4Editor-MyProject2.so
start loading lib file
done loading lib file
pyinit imported
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/DroneLab/UE4PyhtonBridge/src/Wrappers/pyinit.py", line 6, in <module>
    from Wrappers import phandlers
  File "/DroneLab/UE4PyhtonBridge/src/Wrappers/phandlers.py", line 4, in <module>
    import cv2
ImportError: /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0: undefined symbol: cairo_ft_font_options_substitute
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'pyinit' is not defined
[2017.10.04-22.10.11:189][142]PIE: Info Play in editor start time for /Game/Maps/UEDPIE_0_SMALL -60.810588
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'pyinit' is not defined
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'pyinit' is not defined
Traceback (most recent call last):

orig74, 2017-10-04:
It seems that it has some issues with the OpenCV installation.
first to try to install the libpangocairo manually in the unreal docker image by running:
`sudo apt-get install libpangocairo-1.0-0`
If that doesn't help, try to rebuild the image with the nocache flag e.g. 
`cd dockers/unreal_engine_4 && ../build --nocache`

orig74, 2017-10-04:
Please also try installing
`sudo apt-get install libpango1.0-dev`
in the same way before recompiling the image

geevargs, 2017-10-05:
[image]

geevargs, 2017-10-05:
It says project file not found and I can change /DroneLab/MyProject2 but I still get the pyinit not found

geevargs, 2017-10-05:
[image]

geevargs, 2017-10-05:
if i modify the path to the path in the docker and it runs but gives this error and I have already installed  sudo apt-get install libpangocairo-1.0-0 &  sudo apt-get install libpango1.0-dev and recompiled the image but still get the same error

orig74, 2017-10-05:
you need to change the path in the script set_ue4_bridge.sh
and run it again e.g.
`./set_ue4_bridge.sh`

orig74, 2017-10-05:
also, I think I misled you :(
you when you start the unreal docker image you need to give it the project file path on the host machine
e.g.

`PROJECT_FILES_DIR=<your host project file dir (absolute path)> ./run_image.sh
`

orig74, 2017-10-05:
You don't need to reinstall it again. just stop the image and run it again:
`./stop.sh`
`PROJECT_FILES_DIR=<your host project file dir (absolute path)> ./run_image.sh`

geevargs, 2017-10-05:
I just wanna know the Project Dir is from the Host or the docker...it works with the docker

DroneLab$cd MyProject2/
/DroneLab/MyProject2$./build.sh       
ERROR: System.IO.DirectoryNotFoundException: Could not find a part of the path "/home/sujith/DroneSimLab/MyProject2/MyProject2.uproject".
  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, Boolean anonymous, FileOptions options) <0x7f01f5d5d2c0 + 0x001f4> in <filename unknown>:0 
  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, System.String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost) <0x7f01f5d5d210 + 0x0004d> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream:.ctor (string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
  at System.IO.StreamReader..ctor (System.String path, System.Text.Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize, Boolean checkHost) <0x7f01f5bcc530 + 0x000e4> in <filename unknown>:0 
  at System.IO.StreamReader..ctor (System.String path, System.Text.Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize) <0x7f01f5bcc4f0 + 0x00030> in <filename unknown>:0 
  at System.IO.StreamReader..ctor (System.String path, Boolean detectEncodingFromByteOrderMarks) <0x7f01f5bcc420 + 0x0003b> in <filename unknown>:0 
  at System.IO.StreamReader..ctor (System.String path) <0x7f01f5bcc400 + 0x00016> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.StreamReader:.ctor (string)
  at System.IO.File.ReadAllText (System.String path) <0x7f01f5d5b650 + 0x00033> in <filename unknown>:0 
  at UnrealBuildTool.JsonObject.Read (System.String FileName) <0x41c80480 + 0x00014> in <filename unknown>:0 
  at UnrealBuildTool.ProjectDescriptor.FromFile (System.String FileName) <0x41c7f9e0 + 0x00047> in <filename unknown>:0 
  at UnrealBuildTool.RulesCompiler.CreateProjectRulesAssembly (UnrealBuildTool.FileReference ProjectFileName) <0x41c7f4c0 + 0x0005f> in <filename unknown>:0 
  at UnrealBuildTool.UEBuildTarget.CreateTarget (UnrealBuildTool.TargetDescriptor Desc, System.String[] Arguments, Boolean bCompilingSingleFile) <0x41c7e730 + 0x000ab> in <filename unknown>:0 
  at UnrealBuildTool.UnrealBuildTool.RunUBT (UnrealBuildTool.BuildConfiguration BuildConfiguration, System.String[] Arguments, UnrealBuildTool.FileReference ProjectFile) <0x41c77000 + 0x01963> in <filename unknown>:0

orig74, 2017-10-05:
The path needs to be from the docker machine. Since the unreal engine is running under docker, it is not aware of the host file system

geevargs, 2017-10-06:
[image]

geevargs, 2017-10-06:
upgraded unreal from 4.14 to 4.17.2 
now it deosnt get data from sitl

geevargs, 2017-10-06:
[image]

geevargs, 2017-10-06:
I did a ping test from the SITL image to the host works fine...and I did a print after line 100 and before line 83

geevargs, 2017-10-06:
/home/sujith/Downloads/Pasted image at 2017_10_06 01_49 PM.png

geevargs, 2017-10-06:
[image]

geevargs, 2017-10-06:
It not reading the print in line 86

[image]

orig74, 2017-10-07:
You are right , it is a network issue. Did you change the --net=host flag to something else when you run the docker image?

geevargs, 2017-10-07:
No I ran it as is

geevargs, 2017-10-07:
I changed line RUN 22 :  useradd -o -u $UID docker and line 17 I added RUN apt-get -y install software-properties-common python-software-properties

orig74, 2017-10-07:
Did you run the example properly?
you need to run in each tmux pane:
`python drone_main.py`
and the hit the play button in the editor
show me the tmux screen after you do that.

geevargs, 2017-10-07:
[image]

orig74, 2017-10-07:
I am trying to understand if the drones are not moving only in the unreal engine or also in the SITL? do you see the position moves in the tmux windows?

geevargs, 2017-10-07:
It doesn't move in either the SITL or unreal its like it doesn't reach either of them.

orig74, 2017-10-07:
I am suspecting the IP addresses might have changed. Take a look at the config file:
demos/unreal_proxies/two_drones/config.py
make sure the IP addresses there are correct.
This can happen when you have a few docker images running and starting in an order I didn't expect. Make sure that the IP address of zmq_pub_unreal_proxy is indeed 172.17.0.1 if not try and change between them in the config file.

geevargs, 2017-10-09:
1) config.py shows zmq_pub_drone_fdm=('172.17.0.2',5566) and zmq_pub_unreal_proxy=('172.17.0.1',5577) and in ifconfig docker0 is 172.17.0.1.
2) connecting to tcp://172.17.0.3:5566 and tcp://172.17.0.2:5566 in unreal engine docker

orig74, 2017-10-09:
please show me the output of the `ip addr show` of the three containers.

geevargs, 2017-10-09:
[image]

geevargs, 2017-10-09:
I am checking to see if its a firewall configuration and I did a iptables -F and that cleared it up....It works now

orig74, 2017-10-09:
Good!
Let me know if you still have stability issues. The next step will be to try to run it a standalone game

closed 2017-10-13 by orig74