orig74/DroneSimLab #23: Dronemain.py not bieng executed

geevargs, 2017-10-18:
It shows undefined symbol : cairo_ft_font_options_substitute

[image]

orig74, 2017-10-18:
I can see you are using a different version of the Dockerfile, your version starts from ubuntu 16.04 and my version is 14.04 I think this is the reason for the missing lib.

you can try and manually install it before running dronemain.py by running
`sudo apt-get install libpangocairo-1.0-0`
If that works, you can add this line to your DockerFile and build the image.
Let me know how it goes.

orig74, 2017-10-18:
you can also try newer miniconda version and replace the one in your docker file:
this is the list of versions:
[miniconda versions](https://repo.continuum.io/miniconda/)
you can change to the latest one which is:
Miniconda3-4.3.27.1-Linux-x86_64.sh

geevargs, 2017-10-19:
I am getting this error when it tries to install opencv
UnsatisfiableError: The following specifications were found to be in
conflict:
  - opencv3 3.2.0* -> python 3.4* -> xz 5.0.5
  - python 3.6*
Use "conda info <package>" to see the dependencies for each package.

orig74, 2017-10-19:
maybe rolling back to 14.04 in the SItl Dockerfile (like my version) will be the fastest approach. If you want me to check your version. please fork the project (the button on the upper right) do the changes and let me see.

geevargs, 2017-10-19:
I rolled back to 14.04 in the SITL Dockerfile...didnt make a difference.
I basically used the original Dockerfile code without changing the code.

[image]

orig74, 2017-10-20:
I'm rebuilding the SITL image to see if I can recreate the issue.
I'll get back to you...

orig74, 2017-10-20:
I Added this line to the SITL docker file and updated this file and it is working form me:
RUN /miniconda/bin/conda update -y cairo

Let me know how it goes.

geevargs, 2017-10-20:
[image: Inline image 1]

It still deosnt work....

orig74, 2017-10-20:
you didn't paste an image

geevargs, 2017-10-20:
[image]

geevargs, 2017-10-20:
I had to install cairo in the dockerfile otherwise it didnt recognize it.

orig74, 2017-10-20:
The file that worked for me was the one that is currently in the repository. E.g with the Miniconda3-4.2.12-Linux-x86_64.sh version and Ubuntu 14.04. Just take this file as is and run the ../build.py --nocache.

geevargs, 2017-10-20:
That worked but the ardupilot is giving an error now.
[image]

orig74, 2017-10-20:
run this command in the ardupilot pane, so I can see the full log and send it to me
`../Tools/autotest/sim_vehicle.py -w > /dronelab/ardupilot_fail.log 2>&1`
the file will be in the DroneSimLab dir in the host

geevargs, 2017-10-20:
[ardupilot_fail.log](https://github.com/orig74/DroneSimLab/files/1403281/ardupilot_fail.log)

orig74, 2017-10-23:
Hi,
I manage to recreate this error. It was something to do with git relative paths. I updated the build script to deal with this error. Please run the main build script "build.py". It shuld print "updating ardupilot git relative paths" and fix this problem.
Let me know how it goes.

closed 2017-10-24 by orig74