orig74/DroneSimLab #24: failed bulding docker image ros_image_indigo please look at the log file /tmp/dbuild-*.log

jediofgever, 2017-11-04:
So I just followed the set up wiki innocently but when I do `./build.py` the long process end up with the error as in title. Can you give me some hints to overcome this issue.

Thanks.
Atas

orig74, 2017-11-04:
hi,
just attach the files to the comment, so I can have a look. 
ori

jediofgever, 2017-11-05:
Hi, 
Sorry I just see you commented. the output fot ./build.py command is 
"
fet@fet-SATELLITE-C640:~/DroneSimLab$ ./build.py 
skip unreal engine option = False
update submodules...
updating submodule: PX4/Firmware
updating submodule: dockers/unreal_engine_4/UnrealEngine
updating submodule: ardupilot
updating ardupilot git relative paths
updating submodule: UE4PyhtonBridge
downloading baked games...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    48  100    48    0     0     28      0  0:00:01  0:00:01 --:--:--    28
Downloading command: curl -o baked_games/game_demo/game_demo.tgz https://studweb.cosc.canterbury.ac.nz/~oga13/ue4_games//game_demo.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  718M  100  718M    0     0   144k      0  1:24:45  1:24:45 --:--:--  191k
building docker images...
zipping unreal engine might take awhile...
building image  ros_image_indigo
Error:  Error: failed bulding docker image ros_image_indigo please look at the log file /tmp/dbuild-*.log

"
I take a look to the adressed file still didnt understand anythng

orig74, 2017-11-05:
Hi, there supposed to be a log file in the following path:
named:
/tmp/dbuild-ros_image_indigo.log
can you please drag this file to the comment text area. this the output log for each build image.
ori

jediofgever, 2017-11-05:
I just Edited my comment please check my comment above 
thank you very much

orig74, 2017-11-05:
It means that you don't have nvidia card. Do you have nvidia graphic card?
current I am only supporting nvidia drivers.

jediofgever, 2017-11-05:
ı have nvidia card this command will grab associated gpu's `lspci -vnnn | perl -lne 'print if /^\d+\:.+(\[\S+\:\S+\])/' | grep VGA` the output is ;
`01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GT218M [GeForce 315M] [10de:0a7a] (rev a2) (prog-if 00 [VGA controller])`

orig74, 2017-11-05:
please run:

`> nvidia-smi`

& show me the input

jediofgever, 2017-11-05:
I just runned the command `nvidia-smi` output is ;
`NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.`

Appearently I couldnt success to install drivers properly I am installing them again and I will update here as soon as I do

jediofgever, 2017-11-05:
Update: I installed the drivers properly and tried to run `./build.py` again and this time I got bunch of log files in /tmp the log file now becomes ;
```
fet@fet-SATELLITE-C640:~/DroneSimLab$ ./build.py 
skip unreal engine option = False
update submodules...
updating submodule: PX4/Firmware
updating submodule: dockers/unreal_engine_4/UnrealEngine
updating submodule: ardupilot
updating ardupilot git relative paths
updating submodule: UE4PyhtonBridge
downloading baked games...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    48  100    48    0     0     14      0  0:00:03  0:00:03 --:--:--    14
building docker images...
building image  ros_image_indigo
building image  ros_image_kinetic
building image  python3_dev
building image  sitl_image
building image  unreal_engine_4
testing docker images instalation...
Error building image ros_image_indigo Failed, please look at the log file /tmp/dbuild-ros_image_indigo.log
you can try build this image manualy by:
cd dockers/ros_image_indigo && ../build --nocache
```
and the adressed log files I have uploaded them to [here](https://github.com/jediofgever/errors) 
I tried build each or image seperately and tried to debug but I just end up not succesing. Can you please have a look at log files.

orig74, 2017-11-05:
Hi, Ubuntu 16.04 is what I am using so it should work.
can you please run the line:
`cd dockers/ros_image_indigo && ../build --nocache`
this builds the docker ros_image_indigo from scratch. let me know how it goes

jediofgever, 2017-11-06:
Okay so this log file is going to be long sorry for that here is the output;
```
fet@fet-SATELLITE-C640:~/DroneSimLab$ cd dockers/ros_image_indigo && ../build --nocache
building for nvidia driver 340.104
Sending build context to Docker daemon 8.192 kB
Step 1 : FROM ubuntu:14.04
 ---> 3aa18c7568fc
Step 2 : RUN apt-get -y update
 ---> Running in 45552879a490
Get:1 http://security.ubuntu.com trusty-security InRelease [65.9 kB]
Get:2 http://security.ubuntu.com trusty-security/universe Sources [78.6 kB]
Ign http://archive.ubuntu.com trusty InRelease
Get:3 http://archive.ubuntu.com trusty-updates InRelease [65.9 kB]
Get:4 http://security.ubuntu.com trusty-security/main amd64 Packages [852 kB]
Get:5 http://archive.ubuntu.com trusty-backports InRelease [65.9 kB]
Get:6 http://archive.ubuntu.com trusty Release.gpg [933 B]
Get:7 http://archive.ubuntu.com trusty Release [58.5 kB]
Get:8 http://archive.ubuntu.com trusty-updates/universe Sources [243 kB]
Get:9 http://security.ubuntu.com trusty-security/restricted amd64 Packages [18.0 kB]
Get:10 http://archive.ubuntu.com trusty-updates/main amd64 Packages [1287 kB]
Get:11 http://security.ubuntu.com trusty-security/universe amd64 Packages [246 kB]
Get:12 http://security.ubuntu.com trusty-security/multiverse amd64 Packages [4723 B]
Get:13 http://archive.ubuntu.com trusty-updates/restricted amd64 Packages [21.4 kB]
Get:14 http://archive.ubuntu.com trusty-updates/universe amd64 Packages [557 kB]
Get:15 http://archive.ubuntu.com trusty-updates/multiverse amd64 Packages [16.3 kB]
Get:16 http://archive.ubuntu.com trusty-backports/main amd64 Packages [14.8 kB]
Get:17 http://archive.ubuntu.com trusty-backports/restricted amd64 Packages [40 B]
Get:18 http://archive.ubuntu.com trusty-backports/universe amd64 Packages [52.6 kB]
Get:19 http://archive.ubuntu.com trusty-backports/multiverse amd64 Packages [1396 B]
Get:20 http://archive.ubuntu.com trusty/universe Sources [7926 kB]
Get:21 http://archive.ubuntu.com trusty/main amd64 Packages [1743 kB]
Get:22 http://archive.ubuntu.com trusty/restricted amd64 Packages [16.0 kB]
Get:23 http://archive.ubuntu.com trusty/universe amd64 Packages [7589 kB]
Get:24 http://archive.ubuntu.com trusty/multiverse amd64 Packages [169 kB]
Fetched 21.1 MB in 1min 17s (271 kB/s)
Reading package lists...
 ---> d9f5c0b019e0
Removing intermediate container 45552879a490
Step 3 : RUN apt-get install -y software-properties-common
 ---> Running in abebd9e2ed68
Reading package lists...
Building dependency tree...
Reading state information...
The following extra packages will be installed:
  ca-certificates gir1.2-glib-2.0 iso-codes krb5-locales libasn1-8-heimdal
  libcurl3-gnutls libdbus-glib-1-2 libgirepository-1.0-1 libglib2.0-0
  libglib2.0-data libgssapi-krb5-2 libgssapi3-heimdal libhcrypto4-heimdal
  libheimbase1-heimdal libheimntlm0-heimdal libhx509-5-heimdal libidn11
  libk5crypto3 libkeyutils1 libkrb5-26-heimdal libkrb5-3 libkrb5support0
  libldap-2.4-2 libroken18-heimdal librtmp0 libsasl2-2 libsasl2-modules
  libsasl2-modules-db libwind0-heimdal libxml2 openssl python-apt-common
  python3-apt python3-dbus python3-gi python3-pycurl
  python3-software-properties sgml-base shared-mime-info unattended-upgrades
  xml-core xz-utils
Suggested packages:
  isoquery krb5-doc krb5-user libsasl2-modules-otp libsasl2-modules-ldap
  libsasl2-modules-sql libsasl2-modules-gssapi-mit
  libsasl2-modules-gssapi-heimdal python3-apt-dbg python-apt-doc
  python-dbus-doc python3-dbus-dbg libcurl4-gnutls-dev python3-pycurl-dbg
  sgml-base-doc bsd-mailx mail-transport-agent debhelper
The following NEW packages will be installed:
  ca-certificates gir1.2-glib-2.0 iso-codes krb5-locales libasn1-8-heimdal
  libcurl3-gnutls libdbus-glib-1-2 libgirepository-1.0-1 libglib2.0-0
  libglib2.0-data libgssapi-krb5-2 libgssapi3-heimdal libhcrypto4-heimdal
  libheimbase1-heimdal libheimntlm0-heimdal libhx509-5-heimdal libidn11
  libk5crypto3 libkeyutils1 libkrb5-26-heimdal libkrb5-3 libkrb5support0
  libldap-2.4-2 libroken18-heimdal librtmp0 libsasl2-2 libsasl2-modules
  libsasl2-modules-db libwind0-heimdal libxml2 openssl python-apt-common
  python3-apt python3-dbus python3-gi python3-pycurl
  python3-software-properties sgml-base shared-mime-info
[... 353 lines cut ...]
  libpython2.7-stdlib libquadmath0 libsigsegv2 libstdc++-4.8-dev libtiff5
  libtimedate-perl libtsan0 libumfpack5.6.2 libwebp5 libwebpmux1 libx11-6
  libx11-data libxau6 libxcb1 libxdmcp6 libxext6 libxmuu1 linux-libc-dev make
  manpages manpages-dev openssh-client patch python python-chardet
  python-chardet-whl python-colorama python-colorama-whl python-decorator
  python-dev python-distlib python-distlib-whl python-html5lib
  python-html5lib-whl python-imaging python-minimal python-numpy python-pil
  python-pip python-pip-whl python-pkg-resources python-requests
  python-requests-whl python-scipy python-setuptools python-setuptools-whl
  python-six python-six-whl python-urllib3 python-urllib3-whl python-wheel
  python2.7 python2.7-dev python2.7-minimal python3-pkg-resources rsync xauth
0 upgraded, 109 newly installed, 0 to remove and 6 not upgraded.
Need to get 88.5 MB of archives.
After this operation, 264 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  libdpkg-perl dpkg-dev python3-pkg-resources python-chardet-whl
  python-colorama python-colorama-whl python-distlib python-distlib-whl
  python-six python-html5lib python-html5lib-whl python-numpy python-six-whl
  python-urllib3-whl python-requests-whl python-setuptools-whl python-pip-whl
  python-pkg-resources python-urllib3 python-requests python-setuptools
  python-pip python-wheel
E: There are problems and -y was used without --force-yes
The command '/bin/sh -c apt-get -y install python-scipy gawk git python-pip' returned a non-zero code: 100

```

orig74, 2017-11-06:
Hi, 
I can see it fails on step 6 in the docker file and prior to that it fails to update properly the apt repository.
I am currently building again my ros_image_indeo and it looks OK (passed step 6 so far).

have a look at this [link](https://askubuntu.com/questions/29071/apt-get-update-cannot-find-ubuntu-servers) to make sure you don't have network issues.

closed 2017-11-14 by orig74