orig74/DroneSimLab #25: Vehicle in simulation

PujieXin, 2017-11-17:
Is it possible to add vehicles to simulation environment , and  make the Drone track the vehicle(ArduRover2)? If I want add vehicles, which part should I start working on?

orig74, 2017-11-18:
copy the existing demo px4gazebo. 
You will need to add an internal tmux screen which will represent the new vehicle.
the basic structure of the demo is a tmux screen which contains internal running tmux screens for each vehicle and one for the unreal engine.  
You should modify the file runtmux_full_docker.sh this file is running the main tmux.
 in line 27 it runs the script run_tmux_ros_px4.sh which runs the internal tmux of one vehicle.
add after line 27 something like:

```
tmux new-window -n my_vehicle
tmux send-keys "source /DroneLab/scripts/run_tmux_my_vehicle.sh" ENTER
```

the run_tmux_my_vehicle.sh file is the script you need to build in order to make your vehicle work.

another importent thing is to use my version of PX4 which included in the dronesimlab. It is a modified version which sends unreal engine the position of the vehicle.

PujieXin, 2017-11-18:
downloading baked games...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    48  100    48    0     0     39      0  0:00:01  0:00:01 --:--:--    39
building docker images...
zipping unreal engine might take awhile...
building image  ros_image_indigo
Error:  Error: failed bulding docker image ros_image_indigo please look at the log file /tmp/dbuild-*.log

Sorry,here is another question, I was wondering if you could tell me how to slove this problem.Thank you in advance.

PujieXin, 2017-11-19:
I have solved the previous problem. Now I have another problem.
curl: (6) Could not resolve host: studweb.cosc.canterbury.ac.nz
Traceback (most recent call last):
  File "./build.py", line 49, in <module>
    md5=os.popen('curl '+games_path+game_name+'.md5').read().split()[0]
IndexError: list index out of range

It seems that the website is unavailable. How could I solve this problem? Could you please tell me the new website or the method to solve this problem.
Thank you!

orig74, 2017-11-19:
can you check that you can browse:
https://studweb.cosc.canterbury.ac.nz/ ?
it seems online when I tested it.

PujieXin, 2017-11-19:
When I open this page, it shows that:

StudWeb

Welcome to StudWeb, the CSSE student web server. For pages to appear here, place them in your public_html directory. If your usecode is abc987, then files placed in ~abc987/public_html directory will appear on the web here: http://studweb.cosc.canterbury.ac.nz/~abc987/

If you get "permission denied" errors, make sure the files are world-readable: the files themselves should have world-read permissions, while the public_html and home directories must have at least world-execute permissions.

Is this the right status?

orig74, 2017-11-19:
yes, it is where I put the packed games.
try the build script again maybe the site was temporary down

PujieXin, 2017-11-19:
Thank you! I will try it asap!

PujieXin, 2017-11-21:
Some issues when I run the program.Have you ever met these issues before?
[image]
[image]
[image]

orig74, 2017-11-21:
Please pull the latest version and run the build.py again. As it turned out there were some updates in some third parties libraries regarding some parts of the simulation that needed to be updated. Let me know how it goes

closed 2019-11-09 by orig74