orig74/DroneSimLab #26: No Images seen on the image_view windows (px4_gazebo demo)

ykhedar, 2017-12-13:
[image]

The demo starts and drone is also running, it seems. But the ue4_bridge is not publishing any image probably because it is not receiving anything from the game. But I do not understand why. I could provide more information if needed.

ykhedar, 2017-12-13:
I started the game manually and found that it crashes upon start due to some error.
"Could not create OpenGL 3.2 context, Bad Alloc (Insufficient resources for operation.)" following image has the error in more detail.
[image]

orig74, 2017-12-13:
Hi,
Can you please run nvidia-smi in the command line, both for the host machine and in the game guest container (the one that shows the errors). 
paste the output of those commands.

ykhedar, 2017-12-14:
The host machine's output is the following 
`
Thu Dec 14 13:07:11 2017       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 387.34                 Driver Version: 387.34                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  NVS 315             Off  | 00000000:01:00.0 N/A |                  N/A |
| 30%   30C    P8    N/A /  N/A |    160MiB /   963MiB |     N/A      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0                    Not Supported                                       |
+-----------------------------------------------------------------------------+

`

On the guest machine I get this output
`Failed to initialize NVML: Driver/library version mismatch`

Also the guest machine produce the following output for `cat /proc/driver/nvidia/version`
`
NVRM version: NVIDIA UNIX x86_64 Kernel Module  387.34  Tue Nov 21 03:09:00 PST 2017
GCC version:  gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 
`
this output is the same as the host machine..Is it possible the driver is not installed in the guest docker image?

ykhedar, 2017-12-14:
Btw I changed the graphics driver on the host machine a few times because of some other problems. I did this after building the docker images, is it possible that this is the reason for the problem?

orig74, 2017-12-14:
Yes, That's the issue.
when you run the nvidia-smi in the docker container and in the host you should see the same result e.g. the same graphics driver. try to run the build script again.

ykhedar, 2017-12-15:
You are correct. :) Thanks for the help. I am closing this issue.

closed 2017-12-15 by ykhedar