orig74/DroneSimLab #27: Docker for Windows

availn, 2018-01-16:
I'm supposing that this will not run with Docker for Windows without some modifications. Besides handling the graphics drivers and X11, would there be something else that would need to be changed?

orig74, 2018-01-16:
Hi, 
I guess it is possible but it will require a lot of modifications. If you need the host to be windows I think it is worth trying is to use virtual box with pass threw for the graphics card and install Ubuntu 16.04 in VirtualBox and dronesimlab inside it.

closed 2018-01-16 by orig74

R055A, 2018-01-16:
I am using a Windows 10 and Linux Mint 18.3 dual installment. 100 GB of space was required in the /root folder for installing the DroneSimLab framework.