orig74/DroneSimLab #30: Disbale game view window

ykhedar, 2018-03-23:
Hi, is there a way to disable the game view window? I am running the demo - px4_gazebo.

Thanks,  

ykhedar

R055A, 2018-03-23:
Enlarge the other window that you prefer so that it covers the game view window?

ykhedar, 2018-03-23:
Visibility is less of a problem. Since I am not doing anything in this window and it is still being rendered, it is using some computational power (not sure how large or small). So I was wondering if there was a way to switch it off. :)

orig74, 2018-03-25:
If you are using the game mode (packed game) you can change the game parameters in the GameUserSettings.ini file under /LinuxNoEditor/testprj7_14_4/Saved/Config/LinuxNoEditor/. I don't know if you can disable it completely but you can change the screen resolution to something smaller and this maybe will save you some CPU usage. In the Unreal Editor, you can change the rendering parameters and also save some more CPU Usage.

ykhedar, 2018-03-26:
Thanks @orig74 I will try this suggestion.

closed 2018-03-27 by ykhedar