orig74/DroneSimLab #31: Possibility of making the camera look downward for aerial survey

ykhedar, 2018-03-27:
Hi, I have one more doubt. By default the px4_gazebo demo has the camera look away from the ground. Is it possible to make it look on the ground so that an aerial mission can be created. 

Thanks,

khedar

ykhedar, 2018-03-27:
I see there is a topic called "rgb_camera_0down" . I subscribe to it but I see a dark image and not the actual ground surface.

[image]

Any suggestions?

Thanks,

khedar

orig74, 2018-03-27:
I need some more information
what script are you using to run the simulation?

ykhedar, 2018-03-28:
I am using the  "DroneSimLab/demos/px4_gazebo/runtmux_game_docker.sh" script to run the simualtion. I can see that the camera is recording something, i.e. it is moving. But the image is always dark as I pasted in my last comment. 
Any help is greatly appreciated.

Thanks :)

orig74, 2018-03-28:
This demo was created from my unreal engine project and I did initially put two RGB cameras on the drones but when I created the packed game for the demo I only used one of them so I missed the problem with the other one. Maybe it is positioned inside the drone so that's why we see black.  If this feature is important to your research/work please let me know and I will prioritize it.

ykhedar, 2018-03-29:
I would very much like to have it prioritized. This is the only part of my project which is pending at the moment. If you don't have much time or if you could pinpoint the things one needs to do to fix this then I would be happy to do my bit to help. :)
Thanks

closed 2019-11-09 by orig74