orig74/DroneSimLab #32: Commands do not work on my laptop

gonzaloaguilarjimenez, 2018-04-04:
Good afternoon Ori, 

I am Gonzalo Aguilar, doing a PhD on Robotics (Neural Networks) in Spain.

To be able to simulate my NN, I need to combine several programs. The programs to combine are: 
- UE4 
- ROS
- PX4
- AirSim

In the basis of my project i will try to do a simulation of my NN on a SITL on UE4, sensors simulated on PX4 and ROS. Later I will do a real test with a HITL using a real drone but my UE4 environment and finally a test on real environment. 

To check compatibility my Hardware is:
- Intel i7 7700 HQ
- SSR 2050 Mb/s
- Nvidia Gforce GTX 1070

My software is: 
- Ubuntu 16.04.03 LTS

I have already installed by my own and it is already working a combination of: 
- UE4.17.2
- PX4
- XBOX 360 Gamepad
- AirSim

Now I will start explaining my main problems when I try to install DroneSimLab.

[image]

[image]

In these two images we can see: 
1. the result of the docker images after following the process given on DroneSimLab
2. the result when I run the first demo given on Easy Start by running the commands
      $  cd demos/px4_gazebo
      $ ./runtmux_game_docker.sh

I do not really know what the problem is, but I am stucked on this since long time ago and I do not know how to go on.

I hope you can help me.

Thanks in advance.

Best regards.

gonzaloaguilarjimenez, 2018-04-04:
Hello, 

This is what I got after running ./build.py

[image]

Nothing similar...

orig74, 2018-04-04:
Hi,
found the problem
The assert is because the script cant access the games demo path on the web. the game demo is on the university servers and they recently changed the access policy without letting me know. I am working on it now.

orig74, 2018-04-04:
Fixed,
Please pull the latest version, I updated the build script.

gonzaloaguilarjimenez, 2018-04-06:
Hello, 

I was able to install every docker image except that of UE4. I will try it again and let you know.

I wonder if it is possible that I use my own UE4 project with your DroneSimLab App. Could you pleae inform me how to do it? 

Thanks a lot.

Best regards.

orig74, 2018-04-06:
Yes, you can use your project. follow carefully the wiki page [link](https://github.com/orig74/DroneSimLab/wiki/Setup-of-the-Unreal-Engine-4-for-the-demos)
especially the part that initializes the PROJECT_FILES_DIR variable. this is the **parent** folder of your project. This directory will be mapped inside the docker container to `/project_files`  folder.
The make sure you did it correctly after running the UnrealEngine container you can run:
```
cd /project_files
ls
```
you should see you project folder at the output

orig74, 2018-04-06:
As for the unreal engine failure, you can attach the installation log file in:
```
/tmp/dbuild-unreal_engine_4.log
```

R055A, 2018-04-08:
It sometimes doesn't run 100% the first time. It could take more than two runs sometimes in my experiences earlier this year.

There was also a 'ros_image_indigo' that has been removed from the build earlier this year.

You can see it showing in the repository column of the docker images example on the wiki 'Easy start'

gonzaloaguilarjimenez, 2018-04-08:
I have run it multiple times just to check as you said ("could take more than two runs") but still the result is 'not existing the ros image'.

R055A, 2018-04-08:
Maybe you need to reinstall the ros_image_kinetic. I had this issue. It did not install 100% although it showed as installed. To reinstall just kinetic, run '../build' from the kinetic directory.

gonzaloaguilarjimenez, 2018-04-08:
I did as you indicated... as well I installed the ros_indigo image but as you can see there is no good result.  

Besides, the "run_tmux_ros_px4.sh" is shown as not existing...

[image]

orig74, 2018-04-08:
Hi,
Can you explain what is the ros_image_1 folder (under dockers)? 
Did you create it?

gonzaloaguilarjimenez, 2018-04-08:
I did not create it. Previously to installation I deleted all my docker images. 

I think it must be included in the DroneSimLab.

orig74, 2018-04-08:
well, it's not there. have a look at this folder:
[docker folder](https://github.com/orig74/DroneSimLab/tree/master/dockers)

gonzaloaguilarjimenez, 2018-04-09:
Good afternoon again, 

I am trying to reconnect everything with another demo:

[image]

On the image you can see in the terminal top-left containers and images on my dockers, in the terminal to p-right the result when I run both ./run_image.sh and ./attach.sh. Finally center-down terminal you can see the result when I run the : 

~/DroneSimLab/demos/px4_gazebo$ runtmux_full_docker.sh

Several procceses are open in window terminals top-right and center-down but i cannot see any result.

gonzaloaguilarjimenez, 2018-04-16:
Good night again, 

I had to unninstall the docker images and now when I am trying to re-install them again, this is the result with the ROS images:

[image]

After doing everything writen on your site, the result is shown, even after running the command you can see on the last line of the right terminal.

Can you please let me know if something has changed on the server?

Thanks a lot.

Best regards.

gonzaloaguilarjimenez, 2018-04-16:
Maybe this helps: 

[image]

orig74, 2018-04-17:
I am rebuilding this image now. I'll keep you updated

gonzaloaguilarjimenez, 2018-04-18:
Done. 
I will try to run my UE4 project within the next days and inform you.
Thanks a lot.

closed 2019-11-09 by orig74