orig74/DroneSimLab #33: Cannot assign requested address in unreal_proxy.py

geevargs, 2018-04-05:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/airspace/DroneSimLab/UE4PyhtonBridge/src/Wrappers/pyinit.py", line 7, in <module>
    from Wrappers import entry_point
  File "/home/airspace/DroneSimLab/UE4PyhtonBridge/src/Wrappers/entry_point.py", line 4, in <module>
    exec("import "+os.environ['BRIDGE_ENTRY_NAME']+" as main_module")
  File "<string>", line 1, in <module>
  File "/home/airspace/DroneSimLab/demos/unreal_proxies/two_drones/unreal_proxy.py", line 41, in <module>
    socket_pub.bind("tcp://%s:%d" % config.zmq_pub_unreal_proxy )
  File "zmq/backend/cython/socket.pyx", line 547, in zmq.backend.cython.socket.Socket.bind
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc
zmq.error.ZMQError: Cannot assign requested address
Traceback (most recent call last):
  File "<string>", line 1, in <module>
NameError: name 'pyinit' is not defined

geevargs, 2018-04-05:
Even after I assigned the proper address for both FDM and unreal_proxy.py

orig74, 2018-04-06:
try to change the line in the config.py
`zmq_pub_unreal_proxy=('172.17.0.1',5577)`
to 
`zmq_pub_unreal_proxy=('*',5577)`

geevargs, 2018-04-10:
that worked!

orig74, 2018-04-10:
nice!

closed 2018-04-10 by orig74