orig74/DroneSimLab #34: Unreal Simulation Not Working_Command window Running perfect but no display in Unreal Editor

sami8857, 2018-11-09:
If i run only the (./runtmux_game_docker.sh) in px4_gazebo demo It works and show simulation window of gazebo and i can navigate also but the (./runtmux_full_docker.sh) is not working and no unreal engine simulation window opens. Similarly for the other demos underwater and bluerov  (./runtmux_full_docker.sh) it is not working. Everything on the tmux command window seems to be Ok but no unreal simulation window opens
My ultimate goal is to run the demo of underwater and bluerov and further modify them but they are not working. Can you please help regarding that. Following are the screenshots.

For px4_gazebo demo  (./runtmux_game_docker.sh)   This demo is working  fine.

[image]

[image]

For px4_gazebo demo  (./runtmux_full_docker.sh)   This demo is not working. 

[image]

[image]

Similarly others like underwater and bluerov are not working too. No unreal simulation window opens where i can add (PyServer Tick Actor) and it seems to me that project files of demos related to unreal editor are not present because of which no graphical simulation opens in unreal.

For UnderWater demo (./runtmux_full_docker.sh)     Not Working

[image]

[image]

For bluerov (./runtmux_full_docker.sh)               Not Working

[image]

this one even doesnot have the unreal text on the bottom left as in other demos instead 1:bash* is written. Also (./runtmux_game_docker.sh) is also not running for bluerov and same window opens as in this above screenshot.

ThankYou.

orig74, 2018-11-09:
Hi,
Only the ./runtmux_game_docker.sh is currently part of DroneSimLab which is based on a packed game I created from my project. 
The ./runtmux_full_docker.sh is meant to be used as a template to show you how to run when you have an unreal project created.
As for the underwater demo, I'll check.

orig74, 2018-11-09:
In the installation directory run the folowing code to get my version of ardupilot
```
cd ardupilot
git pull https://github.com/orig74/ardupilot.git
``` 
run the demo again

RobertWZ, 2018-12-11:
can this run on ubuntu 18.04?

orig74, 2018-12-12:
haven't tested it on 18.04 but it should work.

closed 2019-11-09 by orig74