orig74/DroneSimLab #35: SSH No GUI

Gilsaa, 2019-02-19:
Hi, I've installed the software on a remote server and want to SSH into this server and see the simulation. However when I run runtmux_game_docker.sh from PX4_Gazebo I don't see any screens. 
The error message says "**X11 connection rejected because of wrong authentication**".

The $DISPLAY variable is set correctly. 
I can use xeyes on the server and the forwarding works. 
I can use xeyes in a simple Docker container and it works. 

Simple Docker - Dockerfile
```
FROM ubuntu
RUN apt-get update && apt-get install -y x11-apps
CMD xeyes
```

Simple Docker - run.sh
```
docker run -it --rm \
--net=host \
--env="DISPLAY" \
--volume="$HOME/.Xauthority:/root/.Xauthority:rw" \
test
```

There seems to be a problem with the X11 forwardning between the Docker host and container. 
Do you have any ideas how to fix this? Or which files to possibly change? I've tried changing the ros_image_kinetic run_image.sh to look more like the simple docker example, but it doesn't work.

Gilsaa, 2019-02-19:
So, I figured it out. 
The .Xauthority file should be put in the home of the user specified in the docker run command, ie. 
--volume="$HOME/.Xauthority:**/home/docker/**.Xauthority:rw. 

Therefore is a user is specified the run command will look like this; 
```
docker run -it --rm --name ros_kinetic \ 
--net=host \
--privileged \
--volume="$DRONE_LAB_DIR:/DroneLab" \
--volume="$DRONE_LAB_DIR:/dockers/docker_home:/home/docker" \ 
--volume="$DRONE_LAB_DIR:/ros/catkin_ws:/ros/catkin_ws" \
--volume="$HOME/.Xauthority:/home/docker/.Xauthority:rw" \
--env="DISPLAY" \
--env="USERNAME=docker" \
--env="USER=docker" \
--env="HOME=/home/docker" \
ros_image_kinetic "bin/bash" $@
```

Now to fix the issues caused by using a nVidia driver instead of the nouveua ...

orig74, 2019-11-09:
Thanks for the update

closed 2019-11-09 by orig74