orig74/DroneSimLab #38: sdl-config is missing while building python3 docker image

lazydroid, 2019-12-20:
Ubuntu 18, nvidia drivers v.430, got this error and no idea how to fix. Installing sdl-config by `sudo apt-get install libsdl2-dev` on the host machine did not change anything.

```
Step 41/42 : RUN PATH=/miniconda/bin:$PATH pip install pygame
 ---> Running in 81819a6dc9b6
Collecting pygame
  Downloading https://files.pythonhosted.org/packages/0f/9c/78626be04e193c0624842090fe5555b3805c050dfaa81c8094d6441db2be/pygame-1.9.6.tar.gz (3.2MB)
    ERROR: Command errored out with exit status 1:
     command: /miniconda/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-grocx2w8/pygame/setup.py'"'"'; __file__='"'"'/tmp/pip-install-grocx2w8/pygame/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-grocx2w8/pygame/pip-egg-info
         cwd: /tmp/pip-install-grocx2w8/pygame/
    Complete output (12 lines):
    
    
    WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
    Using UNIX configuration...
    
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    
    Hunting dependencies...
    WARNING: "sdl-config" failed!
    Unable to run "sdl-config". Please make sure a development version of SDL is installed.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
The command '/bin/sh -c PATH=/miniconda/bin:$PATH pip install pygame' returned a non-zero code: 1
```

lazydroid, 2019-12-20:
ok. changed that by adding at the 25 line of `build` script:

```
### this fixes missing sdl-config during pygame install
RUN apt-get install -y libsdl1.2-dev
```

now the output is:
```
    WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
    Using UNIX configuration...
    
    
    Hunting dependencies...
    SDL     : found 1.2.15
    FONT    : not found
    IMAGE   : not found
    MIXER   : not found
    PNG     : found
    JPEG    : not found
    SCRAP   : found
    PORTMIDI: not found
    PORTTIME: not found
    FREETYPE: found 21.0.15
    Missing dependencies
    ----------------------------------------
```
Would greatly appreciate to get pointers to the missing deps! =)

lazydroid, 2019-12-20:
ok, the correct fix would be adding the following somewhere around line 25 of `build` script:

```
### this fixes missing sdl-config dependencies during pygame install
RUN apt-get install -y libsdl1.2-dev libsdl-image1.2-dev libsdl-ttf2.0-dev libsdl-mixer1.2-dev libportmidi-dev
```

closed 2019-12-20 by lazydroid

orig74, 2019-12-20:
Cheers for that,
I'll add the line to the dockerfile.
Which docker file is it?

lazydroid, 2019-12-20:
dockers/python3_dev/Dockerfile

lazydroid, 2019-12-22:
BTW, are there any plans to change to pygame 1.9.6 (April 25th 2019)?

orig74, 2019-12-23:
Not at the moment.