orig74/DroneSimLab #40: how to edit BlueROV underwater scene?

lazydroid, 2019-12-23:
I'm running BlueROV in simulator, the scenery is nice, but I need to add things ot the scene as well as to the submarine itself. Could you please give me some pointers how this could be done?

Thanks!

orig74, 2019-12-23:
The environment was built using assets bought from the unreal engine marketplace. Specifically, it was the "Ocean Floor Environment". If you buy this environment from the market place (just to fair to the creator), I can send you my unreal project and you can tweak it as you like.

closed 2019-12-23 by orig74

lazydroid, 2019-12-23:
Thank you for all answers, I'll get back to you re: buying the "Ocean Floor".

Cheers,
lenik

lazydroid, 2019-12-24:
I've got the "Ocean Floor" environment, could you please send the project to lenik.terenin@gmail.com ?

Thanks in advance!
lenik