orig74/DroneSimLab #43: Build Failed on Ubuntu 16.04 LTS while building bluerov branch?

adamcropp, 2020-01-02:
osibot@osibot-gigaold:~$ git clone -b bluerov https://github.com/orig74/DroneSimLab.git
Cloning into 'DroneSimLab'...
remote: Enumerating objects: 43, done.
remote: Counting objects: 100% (43/43), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 1442 (delta 15), reused 22 (delta 8), pack-reused 1399
Receiving objects: 100% (1442/1442), 766.00 KiB | 394.00 KiB/s, done.
Resolving deltas: 100% (814/814), done.
Checking connectivity... done.
osibot@osibot-gigaold:~$ ls
Desktop    Downloads    examples.desktop  Pictures  Templates        Videos
Documents  DroneSimLab  Music             Public    terminallog.odt
osibot@osibot-gigaold:~$ cd DroneSimLab
osibot@osibot-gigaold:~/DroneSimLab$ sudo ./build.py --skip_ue4_editor
[sudo] password for osibot: 
skip unreal engine option = True
update submodules...
updating submodule: PX4/Firmware
Submodule 'PX4/Firmware' (https://github.com/orig74/Firmware.git) registered for path 'PX4/Firmware'
Cloning into 'PX4/Firmware'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Total 294639 (delta 3), reused 3 (delta 3), pack-reused 294635
Receiving objects: 100% (294639/294639), 108.48 MiB | 5.37 MiB/s, done.
Resolving deltas: 100% (220385/220385), done.
Checking connectivity... done.
Submodule path 'PX4/Firmware': checked out '050847dae9d266cd348112985e86b5289f9159ba'
Submodule 'NuttX' (https://github.com/PX4/NuttX.git) registered for path 'NuttX'
Submodule 'Tools/gencpp' (https://github.com/ros/gencpp.git) registered for path 'Tools/gencpp'
Submodule 'Tools/genmsg' (https://github.com/ros/genmsg.git) registered for path 'Tools/genmsg'
Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/jMAVSim'
Submodule 'Tools/sitl_gazebo' (https://github.com/orig74/sitl_gazebo) registered for path 'Tools/sitl_gazebo'
Submodule 'cmake/cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon) registered for path 'cmake/cmake_hexagon'
Submodule 'mavlink/include/mavlink/v1.0' (https://github.com/mavlink/c_library_v1.git) registered for path 'mavlink/include/mavlink/v1.0'
Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path 'mavlink/include/mavlink/v2.0'
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/GpsDrivers.git) registered for path 'src/drivers/gps/devices'
Submodule 'src/lib/DriverFramework' (https://github.com/PX4/DriverFramework.git) registered for path 'src/lib/DriverFramework'
Submodule 'src/lib/ecl' (https://github.com/PX4/ecl.git) registered for path 'src/lib/ecl'
Submodule 'src/lib/matrix' (https://github.com/PX4/Matrix.git) registered for path 'src/lib/matrix'
Submodule 'src/modules/uavcan/libuavcan' (https://github.com/UAVCAN/libuavcan.git) registered for path 'src/modules/uavcan/libuavcan'
Submodule 'unittests/googletest' (https://github.com/google/googletest.git) registered for path 'unittests/googletest'
Cloning into 'NuttX'...
remote: Enumerating objects: 17, done.
remote: Counting objects: 100% (17/17), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 469560 (delta 2), reused 7 (delta 1), pack-reused 469543
Receiving objects: 100% (469560/469560), 138.62 MiB | 4.05 MiB/s, done.
Resolving deltas: 100% (360042/360042), done.
Checking connectivity... done.
Submodule path 'PX4/Firmware/NuttX': checked out 'af5d2b058c55431607aa5aa29fc1d1bbc7e51089'
Cloning into 'Tools/gencpp'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 837 (delta 0), reused 3 (delta 0), pack-reused 833
Receiving objects: 100% (837/837), 141.94 KiB | 0 bytes/s, done.
Resolving deltas: 100% (455/455), done.
Checking connectivity... done.
Submodule path 'PX4/Firmware/Tools/gencpp': checked out '26a86f04bcec0018af6652b3ddd3f680e6e3fa2a'
Cloning into 'Tools/genmsg'...
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 1780 (delta 0), reused 1 (delta 0), pack-reused 1769
Receiving objects: 100% (1780/1780), 381.89 KiB | 334.00 KiB/s, done.
Resolving deltas: 100% (947/947), done.
Checking connectivity... done.
Submodule path 'PX4/Firmware/Tools/genmsg': checked out '72f0383f0e6a489214c51802ae12d6e271b1e454'
Cloning into 'Tools/jMAVSim'...
remote: Enumerating objects: 116, done.
[... 415 lines cut ...]
remote: Compressing objects: 100% (4/4), done.
remote: Total 17246 (delta 0), reused 0 (delta 0), pack-reused 17242
Receiving objects: 100% (17246/17246), 5.22 MiB | 1.65 MiB/s, done.
Resolving deltas: 100% (12187/12187), done.
Checking connectivity... done.
Submodule path 'ardupilot/modules/waf': checked out '73163ef5c117a51f29546d17c2bb2041aecefcdd'
updating ardupilot git relative paths
updating submodule: UE4PyhtonBridge
Submodule 'UE4PyhtonBridge' (https://github.com/orig74/UE4PyhtonBridge) registered for path 'UE4PyhtonBridge'
Cloning into 'UE4PyhtonBridge'...
remote: Enumerating objects: 108, done.
Receiving objects: 100% (108/108), 22.65 KiB | 0 bytes/s, done.
Resolving deltas: 100% (47/47), done.
remote: Total 108 (delta 0), reused 0 (delta 0), pack-reused 108
Checking connectivity... done.
Submodule path 'UE4PyhtonBridge': checked out '338f0d85f02e23034fa97541ff05bf6276c00bd0'
downloading baked games...
/bin/sh: 1: curl: not found
Traceback (most recent call last):
  File "./build.py", line 50, in <module>
    md5=os.popen('curl -k '+games_path+game_name+'.md5').read().split()[0]
IndexError: list index out of range
osibot@osibot-gigaold:~/DroneSimLab$ docker images
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
osibot@osibot-gigaold:~/DroneSimLab$

orig74, 2020-01-03:
please install curl and try again:
sudo apt install curl

....& don't build as sudo (it will create problems)

adamcropp, 2020-01-03:
My apologies, I did a fresh install of ubuntu before collecting that last log.
I installed curl and did another build but ran back into the 'original' error with ROS docker image below:

building docker images...
building image  ros_image_kinetic
building image  python3_dev
building image  sitl_image
testing docker images instalation...
Error building image ros_image_kinetic Failed, please look at the log file /tmp/dbuild-ros_image_kinetic.log
you can try build this image manualy by:
cd dockers/ros_image_kinetic && ../build –nocache

/tmp/dbuild-ros_image_kinetic.log
Sending build context to Docker daemon  9.728kB

Step 1/74 : FROM ubuntu:xenial-20170417.1
xenial-20170417.1: Pulling from library/ubuntu
aafe6b5e13de: Pulling fs layer
0a2b43a72660: Pulling fs layer
18bdd1e546d2: Pulling fs layer
8198342c3e05: Pulling fs layer
f56970a44fd4: Pulling fs layer
8198342c3e05: Waiting
f56970a44fd4: Waiting
0a2b43a72660: Verifying Checksum
0a2b43a72660: Download complete
18bdd1e546d2: Download complete
f56970a44fd4: Verifying Checksum
f56970a44fd4: Download complete
8198342c3e05: Download complete
aafe6b5e13de: Verifying Checksum
aafe6b5e13de: Download complete
aafe6b5e13de: Pull complete
0a2b43a72660: Pull complete
18bdd1e546d2: Pull complete
8198342c3e05: Pull complete
f56970a44fd4: Pull complete
Digest: sha256:f3a61450ae43896c4332bda5e78b453f4a93179045f20c8181043b26b5e79028
Status: Downloaded newer image for ubuntu:xenial-20170417.1
 ---> f7b3f317ec73
Step 2/74 : RUN apt-get -y update
 ---> Running in d91d7e5f8422
^[[91mstandard_init_linux.go:211: exec user process caused "exec format error"
^[[0mThe command '/bin/sh -c apt-get -y update' returned a non-zero code: 1
building for nvidia driver 384.130

orig74, 2020-01-03:
hi run:
`cd dockers/ros_image_kinetic && ../build –nocache`
Let me know if it doesn't work and paste the logs.
Just tested it on my side (on the master branch) and it worked fine.

adamcropp, 2020-01-04:
Hi,

When I run the command (I tried it already) I get the same error:
^[[91mstandard_init_linux.go:211: exec user process caused "exec format error"

Glad to hear you have successfully tested it, so the issue is with my setup.
I will try on a few different machines to see if that works...

adamcropp, 2020-01-04:
Before I pull too much of my hair out on this one - as I keep running into different errors...

Do you see any issue with running this on a Amazon AWS EC2 p2.xlarge graphics instance running ubuntu server 18.04 LTS?

Could this software possibly be installed on Windows Server 2016 using docker?

My end goal here is to create a simulated BlueROV (with video stream) that is run on the cloud that people can access via QGroundControl, I just want to make sure I am not barking up the wrong tree...

hassanyf, 2020-01-16:
> hi run:
> `cd dockers/ros_image_kinetic && ../build –nocache`
> Let me know if it doesn't work and paste the logs.
> Just tested it on my side (on the master branch) and it worked fine.

I ran `cd dockers/ros_image_kinetic && ../build –nocache` but I get stuck at "Cloning into opencv". its been over 2 hours, still there.

orig74, 2020-01-18:
please attach the /tmp/dbuild-ros_image_kinetic.log file