orig74/DroneSimLab #45: Wrong place for clang ?

tomlogan501, 2020-03-13:
Hi,
Seems that on Ubuntu, clang is not well extract..

> 

> Step 35/48 : RUN cd /local/UnrealEngine && ./Setup.sh
>  ---> Running in 3f77b5e6ef44
> Fixing inconsistent case in filenames.
> Setting up Mono
> Checking dependencies...
> Updating dependencies: 100% (50690/50690), 9878.0/9878.0 MiB | 12.42 MiB/s, done.
> Result: 0
> Register the engine installation...
> sh: 1: xdg-user-dir: not found
> Installing a bundled clang toolchain
> Downloading toolchain.
> mkdir: cannot create directory '../.git': Not a directory
> /bin/bash: xdg-mime: No such file or directory
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Failed to create the file 
> Warning: ../.git/ue4-sdks/v14_clang-8.0.1-centos7.tar.gz: Not a directory
>   0  693M    0 12888    0     0  25370      0  7:57:38 --:--:--  7:57:38 25370
> curl: (23) Failed writing body (0 != 12888)
> Download failed please fetch from http://cdn.unrealengine.com/Toolchain_Linux/native-linux-v14_clang-8.0.1-centos7.tar.gz manually.

I tried to edit SetupToolchain.sh in Linux directory but still no change

Roman-Malinowski, 2020-04-07:
Hi,
It seems I have the same issue. When looking at _/tmp/dbuild-unreal_engine_4.log_, it raises a lot of warnings in the following steps (I only include one of them but the list is very long)

```
Step 36/48 : RUN cd /local/UnrealEngine && ./GenerateProjectFiles.sh
 ---> Running in 1332dc350b74
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
	updated UE4Printers.py
	no ~/.gdbinit file found - creating a new one.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Generating data for project indexing...
WARNING: Exception while generating include data for UE4Editor: Unable to build: no compatible clang version found. Please run Setup.sh
WARNING: Exception while generating include data for BlankProgram: Unable to build: no compatible clang version found. Please run Setup.sh
```
I guess this is the cause of the final error I get:
```
Step 38/48 : RUN cd /local/UnrealEngine && make
 ---> Running in a8b8aca535ce
bash "/local/UnrealEngine/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Shipping  
Fixing inconsistent case in filenames.
Setting up Mono
Building CrashReportClient...
Creating makefile for CrashReportClient (no existing makefile)
ERROR: Unable to build: no compatible clang version found. Please run Setup.sh
Makefile:207: recipe for target 'CrashReportClient-Linux-Shipping' failed
make: *** [CrashReportClient-Linux-Shipping] Error 5
The command '/bin/sh -c cd /local/UnrealEngine && make' returned a non-zero code: 2
building for nvidia driver 390.132
```

Any idea how I can fix that?

Thanks!

tomlogan501, 2020-04-08:
As a trick, I change the path of extraction of clang in the script files
Contact me if you got other issue.

closed 2020-04-08 by tomlogan501

Roman-Malinowski, 2020-04-08:
Hi Tom,
Thank you for your response. I can't figure out where to change the clang extraction path, could you provide me some insight on the files you modified and the new path you specified?

Thanks

tomlogan501, 2020-04-08:
Hi,
Has mentioned above, it is that file that could be hack : 
DroneSimLab/dockers/unreal_engine_4/UnrealEngine/Engine/Build/BatchFiles/Linux/SetupToolChain.sh

and I edit the TOOLCHAIN_CACHE=../ue4-sdks/

tomlogan501, 2020-04-10:
I create an hotfix based on the 4.23.0 release on my fork 
https://github.com/tomlogan501/UnrealEngine/tree/4.23.0-hotfix

Roman-Malinowski, 2020-04-10:
Thanks a lot!

tomlogan501, 2020-04-14:
@Roman-Malinowski Did you succeed ?

Roman-Malinowski, 2020-04-14:
Hi Tom,

Thank you for asking!
Actually I can't make it work, I have issues when building Unreal with your version. Like before, it delays some install because apt-utils is not yet installed, and the final error I get is the one bellow.
Any idea on how to fix this?
In any case, thank you for your help!

``Setting up vim (2:8.0.1453-1ubuntu1.3) ...
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/vim (vim) in auto mode
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/vimdiff (vimdiff) in auto mode
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/rvim (rvim) in auto mode
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/rview (rview) in auto mode
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/vi (vi) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/vi.1.gz because associated file /usr/share/man/fr/man1/vim.1.gz (of link group vi) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/it/man1/vi.1.gz because associated file /usr/share/man/it/man1/vim.1.gz (of link group vi) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/pl/man1/vi.1.gz because associated file /usr/share/man/pl/man1/vim.1.gz (of link group vi) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ru/man1/vi.1.gz because associated file /usr/share/man/ru/man1/vim.1.gz (of link group vi) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ja/man1/vi.1.gz because associated file /usr/share/man/ja/man1/vim.1.gz (of link group vi) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/man1/vi.1.gz because associated file /usr/share/man/man1/vim.1.gz (of link group vi) doesn't exist
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/view (view) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/view.1.gz because associated file /usr/share/man/fr/man1/vim.1.gz (of link group view) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/it/man1/view.1.gz because associated file /usr/share/man/it/man1/vim.1.gz (of link group view) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/pl/man1/view.1.gz because associated file /usr/share/man/pl/man1/vim.1.gz (of link group view) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ru/man1/view.1.gz because associated file /usr/share/man/ru/man1/vim.1.gz (of link group view) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ja/man1/view.1.gz because associated file /usr/share/man/ja/man1/vim.1.gz (of link group view) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/man1/view.1.gz because associated file /usr/share/man/man1/vim.1.gz (of link group view) doesn't exist
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/ex (ex) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/ex.1.gz because associated file /usr/share/man/fr/man1/vim.1.gz (of link group ex) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/it/man1/ex.1.gz because associated file /usr/share/man/it/man1/vim.1.gz (of link group ex) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/pl/man1/ex.1.gz because associated file /usr/share/man/pl/man1/vim.1.gz (of link group ex) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ru/man1/ex.1.gz because associated file /usr/share/man/ru/man1/vim.1.gz (of link group ex) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ja/man1/ex.1.gz because associated file /usr/share/man/ja/man1/vim.1.gz (of link group ex) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/man1/ex.1.gz because associated file /usr/share/man/man1/vim.1.gz (of link group ex) doesn't exist
update-alternatives: using /usr/bin/vim.basic to provide /usr/bin/editor (editor) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/editor.1.gz because associated file /usr/share/man/fr/man1/vim.1.gz (of link group editor) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/it/man1/editor.1.gz because associated file /usr/share/man/it/man1/vim.1.gz (of link group editor) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/pl/man1/editor.1.gz because associated file /usr/share/man/pl/man1/vim.1.gz (of link group editor) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ru/man1/editor.1.gz because associated file /usr/share/man/ru/man1/vim.1.gz (of link group editor) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/ja/man1/editor.1.gz because associated file /usr/share/man/ja/man1/vim.1.gz (of link group editor) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/man1/editor.1.gz because associated file /usr/share/man/man1/vim.1.gz (of link group editor) doesn't exist
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Removing intermediate container 14ca3f1c9213
 ---> 3069bfa36187
Step 9/48 : RUN curl -o /miniconda.sh https://repo.continuum.io/miniconda/Miniconda3-4.2.12-Linux-x86_64.sh
 ---> Running in 7d12d0e9b7ce
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
[... cut ...]
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
Removing intermediate container 7d12d0e9b7ce
 ---> 20f794075521
Step 10/48 : RUN /bin/bash /miniconda.sh -b -p /miniconda
 ---> Running in 8b334d39b033
Removing intermediate container 8b334d39b033
 ---> 285b52ed7d66
Step 11/48 : RUN PATH=/miniconda/bin conda install -y pyzmq
 ---> Running in 221445d320cf
/bin/sh: 1: conda: not found
The command '/bin/sh -c PATH=/miniconda/bin conda install -y pyzmq' returned a non-zero code: 127``

tomlogan501, 2020-04-15:
Hmm try to delete the docker image to start from new, did you use your own computer or a CI job ?
I have no more errors on the job I deploy but i will verify the image before the demo.
[consoleText.txt](https://github.com/orig74/DroneSimLab/files/4479713/consoleText.txt)