orig74/DroneSimLab #46: Demo ROV

tomlogan501, 2020-04-24:
Hi,
I'm on Ubuntu 19.10, I achieve to create the dockers image and I would like to run the ROV demo.
Seems that some path mismatched to launch the various script in tmux.
I'm on master.

[image]

orig74, 2020-04-24:
Hi,
I am currently working on updating the demo for my research. Due to time constraints I am not supporting previous versions.
Sorry.

tomlogan501, 2020-04-24:
Are you updating the ROV demo or another one ? I'm very interested by your project for our ROV and boat (both on ardupilot)

orig74, 2020-04-24:
Hey,
I am working on simulating our ROV,
You can contact me on oga13@uclive.ac.nz to give me some more details about your work and I'll see what I can do.